kraj/meta-clang #998: Consider move `TUNE_CCARGS:remove = "-mbranch-protection=standard"` to conf file

Oiouuu, 2024-08-13:
Hi, why not place the changes to TUNE_CCARGS in the conf file? 
This change should affect the global CFLAGS. If it's only set in `libcxx_git.bb`, other packages won’t be affected. 
I found that when building other packages, the cxx exception feature doesn’t work properly on aarch64. 
It only works if 
TUNE_CCARGS:remove:toolchain-clang = "${@bb.utils.contains('TUNE_FEATURES', 'aarch64', '-mbranch-protection=standard', '', d)}" 
is added globally.

kraj, 2024-08-14:
https://github.com/kraj/meta-clang/issues/963 reported this issue and fixing C++ runtime was enough to solve that case. However if you are seeing more issues with it then we might want to use it globally infact for all the distros using clang + gcc to build packages will show similar issue on arm64.  Can you add a package where this happens and removing it helps ?

Oiouuu, 2024-08-15:
Here is an example [vsomip](https://github.com/COVESA/vsomeip/blob/master/implementation/configuration/src/configuration_impl.cpp). You can see that Boost is used to load the configuration file with 'try-catch'. Sure I can remove `TUNE_FEATURES` in [vsomeip.bb](https://layers.openembedded.org/layerindex/recipe/401163/), but it's a runtime issue so I can't test all packages.

quaresmajose, 2024-08-19:
I am seeing the same in my distro.
I thought https://github.com/kraj/meta-clang/pull/968 would solve the problem but it didn't and we continue to have problems with the exceptions in runtime.
I will also test https://github.com/kraj/meta-clang/pull/999 to see the result.

We had to revert again https://github.com/foundriesio/meta-lmp/pull/1500 and apply our fix globally 
```
TUNE_CCARGS:remove:toolchain-clang = "${@bb.utils.contains('TUNE_FEATURES', 'aarch64', '-mbranch-protection=standard', '', d)}"
```

Oiouuu, 2024-08-19:
> I am seeing the same in my distro. I thought #968 would solve the problem but it didn't and we continue to have problems with the exceptions in runtime. I will also test #999 to see the result.
> 
> We had to revert again [foundriesio/meta-lmp#1500](https://github.com/foundriesio/meta-lmp/pull/1500) and apply our fix globally
> 
> ```
> TUNE_CCARGS:remove:toolchain-clang = "${@bb.utils.contains('TUNE_FEATURES', 'aarch64', '-mbranch-protection=standard', '', d)}"
> ```

#999 doesn't work, I tried ... :(

quaresmajose, 2024-08-19:
> #999 doesn't work, I tried ... :(

I was hoping it would work, this way it becomes more difficult and perhaps applying the '`-mbranch-protection=standard`' removal on arm64 globally is the most appropriate.

quaresmajose, 2024-08-20:
I have finished my tests and unfortnly the https://github.com/kraj/meta-clang/pull/999 doesn't fix the runtimes exceptions.
I will propose a PR with a global '`-mbranch-protection=standard`' removal for arm64 since the `libcxx` alone is not enough.
However, I don't know which is the most appropriate place for this, would `conf/layer.conf` or `conf/nonclangable.conf` be preferable? looks like the `conf/nonclangable.conf` is more package related.

Oiouuu, 2024-08-23:
#1000

closed 2024-09-02 by Oiouuu

kraj, 2025-07-01:
> Do you know if this is a known upstream clang bug?

I have forgotten the nitty-gritty details but it does have some impact on mixing runtimes between gcc and clang.