kraj/meta-clang #1022: nativesdk-liblldb contains reference to TMPDIR [buildpaths]

sapiippo, 2024-10-18:
Build of nativesdk-clang from meta-clang[master] against poky[styhead] fails with QA error:

```
ERROR: nativesdk-clang-19.1.2-r0 do_package_qa: QA Issue: File /usr/local/oe-sdk-hardcoded-buildpath/sysroots/x86_64-pokysdk-linux/usr/lib/liblldb.so.19.1.2 in package nativesdk-liblldb contains reference to TMPDIR [buildpaths]         
ERROR: nativesdk-clang-19.1.2-r0 do_package_qa: Fatal QA errors were found, failing task.
```

lms-jk, 2024-10-23:
Also affected by this issue.

lms-jk, 2024-10-23:
Adding `LLVM_APPEND_VC_REV=OFF` to `EXTRA_OECMAKE:append:class-nativesdk` in `clang_git.bb` (from https://github.com/kraj/meta-clang/commit/a5ebfbeaec07929844bc902fb386b229e360bde8 ) did not help.

lms-jk, 2024-10-23:
liblldb contains/path/to/TMPDIR/work/x86_64-nativesdk-oesdk-linux/nativesdk-clang/19.1.0/recipe-sysroot-native/usr/bin/python3-native/python3

lms-jk, 2024-10-23:
Setting `LLDB_PYTHON_EXE_RELATIVE_PATH` to `${PYTHON_PN}` fixes the build. However, I am not quite sure if this breaks the Python capabilities of LLDB.

closed 2024-10-24 by kraj