kraj/meta-clang #1074: Both llvm & clang tools installation on target fails

sundeep-kokkonda, 2025-04-02:
Hello,

We are trying to add both llvm & clang tools on target image but that fails with below error.

> ERROR: clang-20.1.1-r0 do_packagedata: Recipe clang is trying to install files into a shared area when those files already exist. The files and the manifests listing them are:
>   .../poky/build/tmp/pkgdata/qemux86-64/runtime-reverse/liblto20.1
>     (matched in manifest-qemux86_64-llvm.packagedata)
> Please adjust the recipes so only one recipe provides a given file.

Steps to reproduce:
1. Clone _poky_
2. Add _meta-clang_ layer
3. Add _IMAGE_INSTALL:append = " llvm clang"_ in local.conf
4. bitbake core-image-minimal

It is because the llvm & clang have same tools and while installing these tools on image has conflicting between the tools. How to handle this? Any CONFLICT/RCONFLICT will solve the issue Or Recipes to be updated? Or Any conditions can be put while installing? Can you suggest a better way.

kraj, 2025-04-03:
I think we  have tried to solve the install conflicts by moving these libraries around, that will be my approach, have same packaging structure in both recipes and then perhaps use one as preferred provider above other. Best approach so far is to let clang provide llvm as well and do not use llvm recipe at all.

kraj, 2025-05-05:
This should be fixed now that clang is merged into core and meta-clang is only a user of it - https://github.com/kraj/meta-clang/pull/1088

Please re-open if issue still persists.

closed 2025-05-05 by kraj