kraj/meta-clang #1083: nativesdk-libclc build error

reatmon-ti, 2025-04-14:
I'm getting some build errors on the latest master when building nativesdk-libclc:

| [3356/3493] cd
/scratch/builds/arago-master-wip/build/build/arago-tmp-default-k3r5/work/x86_64-k3r5-nativesdk-arago-linux/nativesdk-libclc/20.1.2/build
&& prepare_builtins -o clspv64--.bc
/scratch/builds/arago-master-wip/build/build/arago-tmp-default-k3r5/work/x86_64-k3r5-nativesdk-arago-linux/nativesdk-libclc/20.1.2/build/builtins.opt.clspv64--.bc
| FAILED: clspv64--.bc
/scratch/builds/arago-master-wip/build/build/arago-tmp-default-k3r5/work/x86_64-k3r5-nativesdk-arago-linux/nativesdk-libclc/20.1.2/build/clspv64--.bc
| cd
/scratch/builds/arago-master-wip/build/build/arago-tmp-default-k3r5/work/x86_64-k3r5-nativesdk-arago-linux/nativesdk-libclc/20.1.2/build
&& prepare_builtins -o clspv64--.bc
/scratch/builds/arago-master-wip/build/build/arago-tmp-default-k3r5/work/x86_64-k3r5-nativesdk-arago-linux/nativesdk-libclc/20.1.2/build/builtins.opt.clspv64--.bc
| /bin/sh: line 1: prepare_builtins: command not found

Branch: master
Commit: d1e5cbbe1857aa32acc909f54f713b661bcab7d8

kraj, 2025-04-14:
@lumag I think this is due to enabling SPIRV backends

reatmon-ti, 2025-04-14:
I'll do a test build with that target removed from the list.

lumag, 2025-04-24:
@kraj the `prepare_builtins` should be installed by the `clang-native` (or `llvm-native`) recipe. I think I fixed that in OE-Core, but I'm not sure if the fix went in or not.

reatmon-ti, 2025-04-24:
Do you know which patch to OE-Core this was?  Either a link to the message, or even just the subject so that I can search for it?

lumag, 2025-04-24:
@reatmon-ti actually, it was a part of the first libclc commit, see `do_install:append:class-native` in https://git.openembedded.org/openembedded-core/commit/meta/recipes-devtools/llvm/llvm_20.1.2.bb?id=2773404dec7840e2252cfd5bf474be70d8e7888f

reatmon-ti, 2025-04-24:
Ah... this is a nativesdk build, not a native build.  Maybe that is what is missing...

kraj, 2025-05-05:
The changes are merged in core and meta-clang adjustment are merged too - https://github.com/kraj/meta-clang/pull/1088

I think this issue is taken care of, if not please re-open.

closed 2025-05-05 by kraj

reatmon-ti, 2025-05-06:
Last night's build did not fail with this error... it failed with other errors, but those are now for me to look into.  Thanks for fixing this!