kraj/meta-clang #1089: Error:compiler-rt-20.1.2-r0 do_configure:Could not find CMAKE_AR using the following names: x86_64-poky-linux-llvm-ar

harishsadineni1, 2025-04-23:
Hello,

We are trying to build core-image-minimal with bcc, bpftrace packages but that fails with below error.

_Error:
CMake Error at /home/poky/build/tmp/work/core2-64-poky-linux/compiler-rt/20.1.2/toolchain.cmake:11 (find_program):
|   Could not find CMAKE_AR using the following names:
|   x86_64-poky-linux-llvm-ar_

This error is occurring  due to  CMAKE_AR unable to find _x86_64-poky-linux-llvm-ar_ due to broken symbolic link.

> $file x86_64-poky-linux-llvm-ar
> x86_64-poky-linux-llvm-ar: broken symbolic link to ../llvm-ar

Steps to reproduce:

1.Clone poky, meta-clang and meta-openembedded
2.Add meta-clang, meta-openembedded/meta-oe and meta-openembedded/meta-python to bblayers.conf
3. Add the following  in local.conf 
  IMAGE_INSTALL = "bcc bpftool bpftrace"
  IMAGE_FEATURES += "ssh-server-dropbear"
  CORE_IMAGE_EXTRA_INSTALL += "bcc-ptest bpftrace-ptest"

  EXTRA_IMAGE_FEATURES = "allow-empty-password empty-root-password allow-root-login dbg-pkgs dev-pkgs tools-sdk tools-debug tools-profile"
  MACHINE_ESSENTIAL_EXTRA_RRECOMMENDS += "kernel-modules"
  DISTRO_FEATURES:append = "ptest systemd usrmerge"
  VIRTUAL-RUNTIME_init_manager = "systemd"
  DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"

  MACHINE_FEATURES += "hyperv"
  IMAGE_INSTALL:append = " bcc"
  KERNEL_DEBUG:append = "True"
4. bitbake core-image-minimal 

This issue was introduced by the following patch, which removed the LLVM files:
http://lxgit.wrs.com/cgit/layers/meta-clang.git/commit/?h=wr-10.cd-20250414&id=18d270b25e6fc86c2125d7acfe356aca1f8ea4c0

kraj, 2025-04-23:
yeah this is part of meta-clang brokenness introduced with recent mesa 25 upgrade in core. Right now there are patches in progress to merge clang into core layer which should make this situation better.

kraj, 2025-05-05:
Relevant patches are merged and I think it should fix this problem.
Can you test again with latest master oe-core and meta-clang and report back please ?

closed 2025-05-06 by harishsadineni1