kraj/meta-clang #1117: [scarthgap-clang19] Create new branch

adalessandro, 2025-07-21:
Currently working on a chromium `136.0.7103.113` update [0] in scarthgap, which depends on a major clang update to version `v19.1.0`. See build error https://github.com/OSSystems/meta-browser/issues/901#issuecomment-3098058753

This is request to create a new branch in `meta-clang` for this purpose. The new branch should be called `scarthgap-clang19` pulling from the following repository branch:

https://github.com/adalessandro/meta-clang/tree/scarthgap-clang19

[0] https://github.com/OSSystems/meta-browser/issues/901

Regards,
Ariel

adalessandro, 2025-07-21:
@kraj please check https://github.com/OSSystems/meta-browser/issues/901#issuecomment-3098058753 and let me know if you're okay creating this new branch.

adalessandro, 2025-07-22:
@kraj just one missing commit in the branch, see https://github.com/kraj/meta-clang/pull/1118

kraj, 2025-07-23:
> [@kraj](https://github.com/kraj) just one missing commit in the branch, see [#1118](https://github.com/kraj/meta-clang/pull/1118)

yes I was expecting a PR, which you did thanks.

closed 2025-07-23 by kraj