kraj/meta-clang #1121: [scarthgap-clang20] sdk build fails to build libclc

cmeissl, 2025-07-31:
When trying to build a sdk using the `scarthgap-clang20` the build fails on `libclc`.

Error reported by bitbake indicates some issue with `prepare_builtins`:

```sh
...
| /build/tmp-glibc/work/x86_64-nativesdk-av2sdk-linux/nativesdk-libclc/20.1.1/recipe-sysroot-native/usr/bin/opt: WARNING: failed to create target machine for 'spir-unknown-unknown': unable to get target for 'spir-unknown-unknown', see --version and --triple.
| [998/3493] cd /build/tmp-glibc/work/x86_64-nativesdk-av2sdk-linux/nativesdk-libclc/20.1.1/build && prepare_builtins -o clspv--.bc /build/tmp-glibc/work/x86_64-nativesdk-av2sdk-linux/nativesdk-libclc/20.1.1/build/builtins.opt.clspv--.bc
| FAILED: clspv--.bc /build/tmp-glibc/work/x86_64-nativesdk-av2sdk-linux/nativesdk-libclc/20.1.1/build/clspv--.bc
| cd /build/tmp-glibc/work/x86_64-nativesdk-av2sdk-linux/nativesdk-libclc/20.1.1/build && prepare_builtins -o clspv--.bc /build/tmp-glibc/work/x86_64-nativesdk-av2sdk-linux/nativesdk-libclc/20.1.1/build/builtins.opt.clspv--.bc
| /bin/sh: 1: prepare_builtins: not found
...
```

The build output seems to contain the binary `prepare_builtins`, but can't be run on the build host.
Afaict the reason seems to be that it tries to use a non existing interpreter `/usr/local/oe-sdk-hardcoded-buildpath/sysroots/x86_64-av2sdk-linux/lib/ld-linux-x86-64.so.2`

kraj, 2025-07-31:
Can you try this patch to libclc which is proposed from oe-core 

https://patchwork.yoctoproject.org/project/oe-core/patch/20250730061807.1642753-1-raj.khem@gmail.com/

kraj, 2025-07-31:
> The recipe `libclc_git.bb` in the `scarthgap-clang20` branch does not define the function `do_build_prepare_builtins()`. I tried to override the whole recipe with the version from `oe` and add `export LDFLAGS="${BUILD_LDFLAGS}"`, but it seems to result in the same issue.

yeah the problem is that its using nativesdk flags to override it. so this has to be done surgically, perhaps patch the build of libclc