kraj/meta-clang #1132: gcc-runtime still being installed to image

pascalauroboa, 2025-09-18:
When building my image with clang e.g:

local.conf:

```
TOOLCHAIN = "clang"
RUNTIME = "llvm"
TC_CXX_RUNTIME = "llvm"
```

linux-raspberrypi_%.bbappend:
```
DEPENDS:remove:toolchain-clang = " libgcc "

DEPENDS:append:toolchain-clang = " clang-cross-${TARGET_ARCH}"

EXTRA_OEMAKE:append:pn-linux-raspberrypi:toolchain-clang = " LLVM=1"

FILESEXTRAPATHS:prepend := "${THISDIR}/${BPN}:"

SRC_URI:append:toolchain-clang = "\
	file://lto.cfg \
"
```
I would expect that during image install gcc-runtime and libgcc would not be installed onto it. Unfortunately it seems that it is still being installed:
` do_rootfs: Package libgcc cannot be installed into the image because it has incompatible license(s): GPL-3.0-with-GCC-exception`

Is this intended behaviour or is there a setting I missed?

**To Reproduce**
```
Build Configuration:
BB_VERSION           = "2.8.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "centos-9"
TARGET_SYS           = "aarch64-oe-linux"
MACHINE              = "raspberrypi-armv8"
DISTRO               = "ros2"
DISTRO_VERSION       = "xxx"
TUNE_FEATURES        = "aarch64 crc cortexa53"
TARGET_FPU           = ""
DISTRO_NAME          = "Robot Operating System 2 (ROS 2) with OpenEmbedded Linux"
ROS_DISTRO           = "kilted"
ROS_VERSION          = "2"
ROS_PYTHON_VERSION   = "3"
meta-clang           = "scarthgap:057ee563305e9484b29d02347aeafdadc5ea28ed"
meta-networking      
meta-oe              
meta-perl            
meta-python          = "scarthgap:c29a18fa39ede952f3f6108ec007c1906e2d9a0d"
meta                 
meta-poky            
meta-yocto-bsp       = "scarthgap:9ce8f26d37c2e3e0030fd1f36f5c54209c269a05"
meta-raspberrypi     = "scarthgap:aaf976a665daa7e520545908adef8a0e9410b57f"
meta-ros-common      
meta-ros2            
meta-ros2-kilted     = "scarthgap:bbf7d8db818ab40b8e9ed10cf702bbae5ac7f2cd"
```
 Relevant settings for local.conf see above.

If you need any logs or task-depends.dot files please let me know.

kraj, 2025-09-19:
@pascalauroboa yeah, we currently do have some dependencies left w.r.t runtime, some packages do pull GCC runtime.  I am wondering whats the reason for `compiler-rt -> gcc-runtime` dependency, I think at one point this was not a dependency but things might have changed or crept in, I think we can solve it. Could you dig a bit into this and zoom a bit on whats bringing this dependency into chain.

kraj, 2025-09-22:
Yeah such things can cause these sort of issues. We need to understand the reason why we are doing this, I think the reason this is needed is that glibc needs unwinding support in libgcc for this context. Perhaps an acceptable alternative is to avoid that or have an option to avoid it so we can avoid libgcc dependency.

pascalauroboa, 2025-09-22:
Both of the python packages had cargo dependencies. With the bbappend from above both of these dependencies vanished as well. perl-modules is not directly pulling in libgcc and is also not installed on the system. The reason for including libgcc is linked above the recipe. 

https://bugs.python.org/issue37395
https://github.com/python/cpython/issues/88600

It seems that in newer versions it has been fixed but I still need to test it.

pascalauroboa, 2025-09-24:
I used scarthgap and walnascar. I have not tried master. I think the fix for libstd-rs could possibly be added as a bbappend. The python one is a bit more difficult since its more a runtime dependency of glibc which python can trigger via a call to thread_exit.

pascalauroboa, 2025-10-08:
This bbappend builds compiler-rt without gcc:
```
DEPENDS:remove:class-target = " gcc-runtime "
DEPENDS:remove = "libgcc"
DEPENDS:append:class-target = " libcxx-native compiler-rt-native"

# UNWINDLIB:class-nativesdk:toolchain-clang = ""
# COMPILER_RT:class-nativesdk:toolchain-clang = ""
# LIBCPLUSPLUS:class-nativesdk:toolchain-clang = ""
# UNWINDLIB:class-native:toolchain-clang = ""
# COMPILER_RT:class-native:toolchain-clang = ""
# LIBCPLUSPLUS:class-native:toolchain-clang = ""
UNWINDLIB:class-target:toolchain-clang = ""
COMPILER_RT:class-target:toolchain-clang = ""
LIBCPLUSPLUS:class-target:toolchain-clang = ""

EXTRA_OECMAKE:append:class-target = " \
  -DCMAKE_C_COMPILER_WORKS=1 \
  -DCMAKE_CXX_COMPILER_WORKS=1 \
  -DCOMPILER_RT_BUILD_BUILTINS=ON \
  -DCOMPILER_RT_BUILD_ORC=OFF \
"

```

kraj, 2025-10-08:
> This bbappend builds compiler-rt without gcc:
> 
> ```
> DEPENDS:remove:class-target = " gcc-runtime "
> DEPENDS:remove = "libgcc"
> DEPENDS:append:class-target = " libcxx-native compiler-rt-native"
> 
> UNWINDLIB:class-target:toolchain-clang = ""
> COMPILER_RT:class-target:toolchain-clang = ""
> LIBCPLUSPLUS:class-target:toolchain-clang = ""
> 
> 
> EXTRA_OECMAKE:append:class-target = " \
>   -DCMAKE_C_COMPILER_WORKS=1 \
>   -DCMAKE_CXX_COMPILER_WORKS=1 \
>   -DCOMPILER_RT_BUILD_BUILTINS=ON \
>   -DCOMPILER_RT_BUILD_ORC=OFF \
> "
> ```

This change look a good start.

kraj, 2025-10-18:
> libstd-rs per default also uses libgcc for stack unwinding. By setting
> 
> ```
> CARGO_FEATURES:toolchain-clang += " \
>   system-llvm-libunwind \
> "
> ```
> 
> In a bbappend it uses libunwind instead.

a hack would be something like `echo "INPUT(-lunwind)" > /usr/lib/libgcc_s.so`